pcb2gcode/pcb2gcode #673: Don't compile from source for debian 12

alphaonex86, 2023-06-05:
## Description of bug:

## What did you try (include command-line arguments):
1. autoreconf
2. ./configure

## What happened:
configure: error: Package requirements (gdkmm-2.4 >= 2.8) were not met:
Package 'gdkmm-2.4', required by 'virtual:world', not found
then request GUI

## What did you expect to happen:
Work in CLI only on headless server

## pcb2gcode version (run pcb2gcode --version to see this):
git version

eyal0, 2023-06-07:
Can you install that missing package?

alphaonex86, 2023-06-08:
not the libgerbv-dev ... missing into debian 12

ownedcore, 2025-01-07:
You don't need libgerbv-dev. If **./configure** bails on:

`configure: error: Package requirements (libgerv >= 2.1.0) were not met:`

Edit the pkgconfig file (**/usr/lib/x86_64-linux-gnu/pkgconfig/libgerbv.pc)** of the gerbv package. Check the line starting with "Version". It has no value. Just add "2.7" after the colon to have this complete line:

`Version: 2.7`

then run ./configure again.

Funny enough even the **libgerbv.cp** file in debian testing (trixy) is still missing the version value.

eyal0, 2025-01-08:
It's a gerbv bug. The build uses git to get the version number and if git is missing, it doesn't output any version number.

ownedcore, 2025-02-09:
I just find it kinda funny that its still in there.

As I fiddled around compiling pcb2gcode with geos on deb 12 and for easing my mind, I now have a tested method for it. At least on my daily machine and on a fresh install of deb 12 in console mode only. For anyone interested here are the steps.

Starting from a fresh install of debian bookworm (deb 12). Mostly following the route of the Readme step "Installation from git".

`apt install build-essential automake autoconf autoconf-archive libtool libboost-program-options-dev libgtkmm-2.4-dev gerbv git librsvg2-dev`

Now fix the gerbv bug and add version number of "2.9.6" to file **/usr/lib/x86_64-linux-gnu/pkgconfig/libgerbv.pc** on the line starting with _Version_ (Add the number after the colon (:)).

Next we add libgeos++-dev package

`apt install libgeos++-dev`

Clone the repository and change to directory

`git clone https://github.com/pcb2gcode/pcb2gcode.git`
`cd pcb2gcode`

Now we start compiling following the Readme

`autoreconf -fvi`
`./configure`
`make`

Which leaves me with the following output of **./pcb2gcode --version**

> 2.5.0
> Git commit: latest
> Boost: 107400
> Gerbv: 2.7
> Geos: 3.11.1

Test output with a "F_Cu.gbr" from Kicad looks nice to me eyes.
Hope it helps or at least not bothers.

eyal0, 2026-04-05:
pcb2gcode treats warnings as errors and some versions of g++ report errors even if they are in boost libraries.

The correct solution is to pass `-DPCB2GCODE_COMPILE_WARNING_AS_ERROR=OFF` to `pcb2gcode`.

g++-10 does not have this problem.  I think that clang++ doesn't have it either.

let me know if you have more problems.

closed 2026-04-05 by eyal0