pcb2gcode/pcb2gcode #675: Feature Request: export each tool to its own file

sybrenstuvel, 2023-07-14:
## Description:

When milling a PCB, I want to use various tools to remove the copper: 0.1mm and 0.4mm V-bits. Because they're mounted by hand, they'll be in slightly different positions every time. For consistent results, a Z-probe is necessary between tool changes. When the GCode file has all the code in sequence, my CNC mill pauses for tool changes and doesn't allow movement, let alone a Z-probe. Because of this, I would like to have more control over which tool gets into which file.

## pcb2gcode version (run pcb2gcode --version to see this):

```
$ pcb2gcode.exe --version
2.5.0
Git commit: latest 
Boost: 108100      
Gerbv: 2.9.6~77211d
Geos: Not installed
```

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**  You might need to zip them.  If you don't provide a millproject file, make sure that you copy-paste the command-line that you used.

Here is a set of example files. It contains `Gerber_koelbox-controller-gerber/Gerber_TopAssemblyLayer.GTA`,  with has some copper removal for text & removing a bigger rectangle (clearing the copper underneath an antenna). It's converted to GCode in `Gerber_koelbox-controller-gcode/Gerber_TopAssemblyLayer.ncc`.

[gerber-and-gcode.zip](https://github.com/pcb2gcode/pcb2gcode/files/12052759/gerber-and-gcode.zip)

This is the command I used to generate the GCode from the Gerber:

```
pcb2gcode
  --front Gerber_koelbox-controller-gerber/Gerber_TopAssemblyLayer.GTA 
  --front-output Gerber_TopAssemblyLayer.ncc
  --mill-feed 70
  --mill-speed 20000
  --zwork -0.065
  --milling-overlap 25%
  --invert-gerbers
  --isolation-width 1000
  --mill-diameters 0.500,0.200
  --zchange 22.000
  --metric
  --metricoutput
  --tolerance 0.01
  --nog64
  --nom6
  --zsafe 1.000
  --spinup-time 0.5
  --path-finding-limit 1000
  --output-dir Gerber_koelbox-controller-gcode
```

What I would love it an extra option like `--split-by-tool` that exports the GCode for each tool to `Gerber_koelbox-controller-gcode/Gerber_TopAssemblyLayer-T{N}.ncc`

eyal0, 2023-07-14:
The generated gcode has within it the instructions to tell the machine to do a tool change.  It's expected that you machine will see the tool change instruction and then either do it by itself or you will, by hand, put in the new tool and run the probing.  And then you push the play button on your CNC and the gcode will continue to excute.

Are you saying that, for your system, this is not working?  What CNC do you have?

sybrenstuvel, 2023-07-16:
What you're describing is indeed what I, a CNC novice, expected. I guess my machine is mocking me. When it's pausing on the `M5 (Spindle stop.)` it it's literally ignoring everything until I un-pause, and then the program simply continues running.

I have a [SainSmart Genmitsu PROVer v2](https://www.sainsmart.com/products/genmitsu-3018-prover-v2-upgraded-semi-assembled-cnc-router-kit), and I'm using [Candle](https://github.com/Denvi/Candle) to control it.

sybrenstuvel, 2023-07-18:
I'll give bCNC another try. Last time I tried to use it it was spewing loads of uncaught exceptions in the terminal, which didn't instill much confidence. But, what you say sounds good.

wryan67, 2023-07-22:
Each of my tools needs a different milling speed.  I didn't see an option for that in pcb2gcode, so I just wrote a shell script that calls pcb2gcode for each tool, with different parameters on each pass.   That way I also get multiple files, which I prefer over the pausing that is done when you specify all the tools at once.  It just less confusing that way and if the power goes out, restarting is much easier.

c-, 2023-07-22:
>Each of my tools needs a different milling speed. I didn't see an option for that in pcb2gcode

#470 was intended to address tool-specific parameters.

wryan67, 2023-07-22:
Sure, #470 has some good ideas, but that syntax is pretty convoluted.  I wasn't complaining, and am actually pretty happy the way it is.  I was more just trying to explain that using a shell script is a great way to create separate files because you can then customize each pass to your hearts content, and repeat as needed since your options are now scripted.  :)

sybrenstuvel, 2023-07-22:
I didn't get around to testing bCNC properly. In the mean time, I've written a tool that can split the output of PCB2GCode into a separate file per tool. It's part of my toolset at https://gitlab.com/dr.sybren/gcodetool

Basically all you need is to run `gcodesplit thefile.ncc` and it'll create `thefile-mill diameter 0.20000mm.ncc` etc.