pcb2gcode/pcb2gcode #677: When zsafe and zwork are both zero, the gcode file does not match the silkscreen. The output is correct if I change both to another value, say 0.1.

eyal0, 2023-08-21:
When zsafe and zwork are both zero, the gcode file does not match the silkscreen. The output is correct if I change both to another value, say 0.1.

For example, with zsafe = zwork = 0:
G00 X2.09064 Y34.89830 ( rapid move to begin. )
G01 F100.00000
G04 P0 ( dwell for no time -- G64 should not smooth over this point )
G00 Z0.00000 ( retract )

Same code section generated with zsafe = zwork = 0.1:
G00 X2.09064 Y34.89830 ( rapid move to begin. )
G01 F100.00000
( Mill infeed pass 1/1 )
G01 Z0.10000
G04 P0 ( dwell for no time -- G64 should not smooth over this point )
G01 F200.00000
G01 X2.09064 Y34.89830
G01 X2.09064 Y36.22832
G01 X3.42066 Y36.22832
G04 P0 ( dwell for no time -- G64 should not smooth over this point )
G00 Z0.10000 ( retract )

Is this a bug, or am I missing something?

_Originally posted by @hustav in https://github.com/pcb2gcode/pcb2gcode/issues/72#issuecomment-1631788106_

closed 2023-08-21 by eyal0

RWL-69, 2025-04-02:
Hello, this is my first day testing pcb2gcode for leaser etching with a Grbl 1.1 machine.  Grbl uses M3 as a constant laser power and M4 for dynamic laser power.

Just did a test converting the M3 in the generated ngc file to M4 and the laser worked as I expected.  Many G04 P0 commands in the gcode caused the laser to come on and burn more at this point.   A G04 P0 is at the beginning of every etch start affecting the quality of the etch.  G04 P0 is a dwell time required for CNC milling but not for laser.

A laser flag would be an option. 
I will add more comments as I test further.

eyal0, 2025-04-02:
If you find a solution that allows you to fix the gcode with search and replace than I suggest you just stick with that.  I probably won't add another flag for stuff that can be done easily by search and replace because there's just too many of them.

There are users that use lasers run by grbl, like you. There are users that use lasers run by Marlin. There are CNC users on grbl and Marlin, too. And even more combinations with auto leveling, MACH codes, and more. It's a lot! There are probably a dozen flags to handle all the various requests. You can look through the wiki to see some of them.

So if there's something that absolutely has to be in pcb2gcode because it alters the functionality then okay. But anything that can be solved easily with an external script should just keep doing that. There are already too many flags.

PR #678 (2023-08-21): fix: Always perform at least one step of milling.
This fixes #677