pcb2gcode/pcb2gcode #683: additional artificial traces

gx-turbo, 2023-11-14:
## Description of bug:

## What did you try (include command-line arguments):
generate a cnc file from gerber with pcb2gcode, see attached zip.

## What happened:
.cnc and .svg are created with with artificial traces. 
there are additional traces in traced_back_final.svg, but not in traced_back.svg.

## What did you expect to happen:
create correct cnc

## pcb2gcode version (run pcb2gcode --version to see this):
2.5.0

[Pendant-Arduino.zip](https://github.com/pcb2gcode/pcb2gcode/files/13354933/Pendant-Arduino.zip)

eyal0, 2023-11-14:
Can you please read the page on optimizations and let me know if what you are seeing it one of those?

It's in the wiki.

eyal0, 2023-11-14:
What you are seeing as artificial traces might just be the software trying to save your time.

gx-turbo, 2023-11-15:
thank you eyal0,
I set backtrack=0 and the extra traces are gone!
kind regards, Gerd

eyal0, 2023-11-16:
Can you tell me: did the backtrack cause the board to no longer function properly? Because the goal of backtrack is to make a board that will take less time to mill yet still create a board with the correct function. Like, it shouldn't cut any traces, it just does some shortcuts during the milling so that your milling is faster.

If you find that backtrack made a slower milling or an incorrect board then I'd need to fix that. If you have images that you can upload of before and after, that saves me the trouble of having to run your design! That would help me and I would appreciate it

gx-turbo, 2023-11-16:
I don`t think that backtrack made an incorrect board. My first post has a zip attached WITH backtrack, now I attach the result with backtrack=0 and optimize=0.02. back.ngc is reduced from 77k to 46k and the additional traces are gone.
You can compare the svg in both zips.

One additional comment to the wiki: the text below --optimize says:  `Default is "true"`. This might be wrong, because optimize expects a numerical value, which could be somehow the allowed deviation against the exact path.
Maybe this can be clarified.

[Pendant-Arduino-no-backtrack.zip](https://github.com/pcb2gcode/pcb2gcode/files/13378193/Pendant-Arduino-no-backtrack.zip)

eyal0, 2023-11-16:
Thanks for that!  Using the [`image-diff-gui`](https://github.com/eyal0/image-diff-gui/tree/main) tool I can see the difference is this:

[image]

The difference is that rather than lift the tool to get from one trace to another, pcb2gcode chose to just mill over there.  It's not cutting through any traces so it shouldn't affect your design at all.  However, if you were counting on that area to be unmilled, like, you're using it as a copper pour, then it's going to affect that!  I would suggest that if you *really* want a copper pour, do one in KiCad!

[image]

Here's your milling.  Above is with the backtrack.  I circled some of the solid lines showing where we will be milling.  Below is without backtrack.  I circled the dotted lines where the milling tool will go up, across, and back down.  You can see in the stats that the top one does less rapid movement (dotted lines 589.821mm) and less motion overall (3870.97mm).  But the lower one does more rapid movement (799mm) and more motion overall (3956.25mm).  The timings look nearly identical but I think that it because bCNC is not calculating it correctly.  `pcb2gcode` does it quite accurately based on these settings:

```
  --g0-vertical-speed arg (=0.0211667 m s^-1)
                                        speed of vertical G0 movements, for use
                                        in path-finding
  --g0-horizontal-speed arg (=0.0423333 m s^-1)
                                        speed of horizontal G0 movements, for 
                                        use in path-finding
```

Using those, it can figure out if it's faster to mill across or to go up, across, and back down.  And if milling it faster, it will do milling.

optimize default is 0.1 mils.  Changing it will cause gcode to be shorter but the circles will appear less round and more blocky.

closed 2026-11-13 by eyal0