pcb2gcode/pcb2gcode #686: [Question] What should adjacent/overlaping regions on cut layer be interpreted as?

InnerBushman, 2024-05-03:
When trying to use two footprints that have geometry on the same layer designated by user to be cut as an edge cut, the behavior is inconsistent as you can see in the SVG file:
[image]

I'm curious what should be the proper interpretation of adjacent and/or overlapping regions on such layers? Should this be reported as a bug? Just as a side-note I've uploaded the gerber files to PCBway to see their interpretation and the preview showed completely overlapping and adjacent regions as continuous, and partially overlapping as "exclusive or".

I'm attaching 4 gerber examples:
[Adjacent_regions.zip](https://github.com/pcb2gcode/pcb2gcode/files/15198946/Adjacent_regions.zip)
[Continuous_region.zip](https://github.com/pcb2gcode/pcb2gcode/files/15198948/Continuous_region.zip)
[Overlaping_regions.zip](https://github.com/pcb2gcode/pcb2gcode/files/15198949/Overlaping_regions.zip)
[Partialy_overlapping_regions.zip](https://github.com/pcb2gcode/pcb2gcode/files/15199324/Partialy_overlapping_regions.zip)

eyal0, 2024-05-03:
You didn't send any gerber files so it's hard for me to debug this issue.

Please [read about edge cuts](https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Outline#--fill-outline) in the wiki.  If it solves your issue, please resolve this.  If not, then please send me a copy of the gerber files that are problematic and I will take a look.

InnerBushman, 2024-05-03:
> You didn't send any gerber files so it's hard for me to debug this issue.

Sorry, my bad. Forgot to add the gerbers to the zip files. I've updated my post.
I've read the wiki and I presumed the regions must be continuous. But I would be grateful for clarification. Technically both shapes in my project create a loop, but in my case one shape shares edges with the other (a hole for a key switch and a hole for key cap support, separate footprints). 

Thank you for the quick reply.

eyal0, 2024-05-30:
Can you send me the command line that you used, too?

I see a single gerber file and no command line.

eyal0, 2024-05-30:
This is the correct format for a bug report:

https://github.com/pcb2gcode/pcb2gcode/blob/master/.github/ISSUE_TEMPLATE/bug_report.md

I need the command line that you used in order to run pcb2gcode along with all the input files to pcb2gcode.  It's also not clear to me what you are expecting as output from pcb2gcode.  But I will try my best.

The gbr file looks like this:

[image]

If I run this command:

```
~/private/pcb2gcode/pcb2gcode --outline="test01-User_Eco1.gbr" --metric=true --metricoutput=true --mirror-axis=0.0000 --mirror-yaxis=false --nog64=true --optimise=0.0100 --tile-x=1 --tile-y=1 --tolerance=0.0060 --zchange=25.0000 --zchange-absolute=false --zero-start=true --zsafe=1.0000 --extra-passes=2 --mill-feed=200 --mill-speed=7000 --offset=0.0200 --voronoi=false --zwork=-0.0250 --drill-feed=50 --drill-side=auto --drill-speed=7000 --milldrill=false --nog81=true --nog91-1=true --onedrill=false --zdrill=-1.5000 --bridges=0.5000 --bridgesnum=6 --cut-feed=12 --cut-infeed=0.6000 --cut-side=auto --cut-speed=7000 --cut-vertfeed=50 --cutter-diameter=1.1500 --zbridges=-0.5000 --zcut=-1.2000
```

Then my resultant ngc file is this:

[image]

It's cutting along the lines provided.  Is that what you want?

If it's two shapes that share an edge and you make this the edge cut layer, it ought to cut along that edge and work fine, right?  Yes, there will be some overlap, so it'll just recut something that was already cut.  No harm?