pcb2gcode/pcb2gcode #687: Invalid/extra cut breaks board.

cacheflood, 2024-07-18:
## Description of bug:
Facing this issue often found that path-finding-limit value other than 0 causes invalid cuts.

fails:
## What did you try (include command-line arguments):
millproject without path-finding-limit=0 value
## What happened:
Broken / invalid cuts over pcb board.

ok:
## What did you try (include command-line arguments):
millproject with path-finding-limit=0 
## What happened:
board ok.

## pcb2gcode version (run pcb2gcode --version to see this):
2.5.0
Git commit: latest-dirty
Boost: 108300
Gerbv: 2.9.8
Geos: 3.12.

Linux -------- 6.9.8-200.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Jul  5 16:20:11 UTC 2024 x86_64 GNU/Linux

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**  You might need to zip them.  If you don't provide a millproject file, make sure that you copy-paste the command-line that you used.

[working.zip](https://github.com/user-attachments/files/16288715/working.zip)

[notworking.zip](https://github.com/user-attachments/files/16288717/notworking.zip)
[image]
[image]

eyal0, 2024-07-25:
I'm seeing the same error as you.  I worked on it for a while but no success yet.  It seems that the path finder is sometimes finding invalid paths.

cacheflood, 2024-10-07:
Seems that offset setting affects this same case. Attached example uses offset=0.1, but changing it to f.e. 0.2 works ok.
(marked millproject file with line # ######## BUG   ############## CHANGE offset=0.2 => works ########################)

[image]
[millproject-cacheflood.tar.gz](https://github.com/user-attachments/files/17282715/millproject-cacheflood.tar.gz)

eyal0, 2025-12-24:
Possibly fixed in #743. Reopen this if the bug is not fixed.

closed 2025-12-24 by eyal0