pcb2gcode/pcb2gcode #688: Nonsense milling path on a specific isolation width

Lycowolf, 2024-08-03:
## Description of bug:
pcb2gcode generates a nonsensical milling path that goes way outside the outline. It happens when I set isolation width to 0.7 (see below). Mesing with optimization options didn't have any effect.

## What did you try (include command-line arguments):
Gerber file is attached. It was generated using KiCAD.

```shell
pcb2gcode --config=../../../millproject,millproject --mill-diameters=0.5 --bridgesnum=3 --isolation-width=0.7
```

```ini
cat ../../../millproject                                                                                                                                                                                                      
# general
metric=true
zero-start=true
metricoutput=true

# compatibility with GRBL
nog64=true
nog81=true

# milling
mill-diameters=0.2
zwork=-0.1
zsafe=1
mill-feed=200
isolation-width=0.8
mill-speed=10000

# drilling
zdrill=-1.9
drill-speed=10000
drill-feed=50
# G-code senders misinterpret G53, WTF
zchange-absolute=false
zchange=10
onedrill=true

### outline
cutter-diameter=0.8
zcut=-1.9
cut-feed=120
cut-vertfeed=20
cut-infeed=1
cut-speed=10000
# we are single-sided
cut-front=true
bridges=1
bridgesnum=4
zbridges=-0.8

# optimization
#path-finding-limit=50000
# workaround for https://github.com/pcb2gcode/pcb2gcode/issues/687
path-finding-limit=0
```

```ini
cat ./millproject                                                                                                                                                                                                             
front=pcb-F_Cu.gbr
front-output=front.nc
![processed_front_final_0 6](https://github.com/user-attachments/assets/dab2be34-ea65-4c4c-808c-131bb585c197)

outline=pcb-Edge_Cuts.gbr
outline-output=outline.nc
drill=pcb-PTH.drl
drill-output=drill.nc
```

## What happened:
See attached SVGs. The nonsense path is clearly visible

## What did you expect to happen:
Similar result as when isolation-path=0.6 is set.

## pcb2gcode version (run pcb2gcode --version to see this):
```shell
pcb2gcode --version                                                                                                                                                                                                         
2.5.0
Git commit: latest
Boost: 108300
Gerbv: 2.10.1-dev~ccf6a3
Geos: 3.12.2
```

[image]
[image]

Lycowolf, 2024-08-03:
I don't know what happened, but it seems GitHub mangled the SVGs. Originals are in the attached ZIP

[processed_front_final-0.x.zip](https://github.com/user-attachments/files/16480643/processed_front_final-0.x.zip)

eyal0, 2024-08-22:
I don't see the gerber files.  Can you add those?

eyal0, 2024-08-22:
That extra path seems to run right along the voronoi barrier between traces.  Strange.  I'm not sure what causes that.

Lycowolf, 2024-08-22:
Sure. I Hope I included everything; sorry, I am still quite new to PCB design and manufacturing.

[gerbers.zip](https://github.com/user-attachments/files/16719216/gerbers.zip)

eyal0, 2024-08-22:
When I run pcb2gcode with 0.6 and I compare to the 0.6 image, I see no differences at all:

[image]

However, when I run with isolation width 0.7 and compare to your 0.7 image provided, it is different:

[image]

And it's not too too different, just in that one weird move.  I can see that the 0.7 worked right because, when I compare 0.7 against 0.6, the diff is much greater.

[image]

So it seems like maybe something it going wrong but only on your end...?

Maybe we're on different versions of pcb2gcode and that's the issue?  How did you download and run pcb2gcode?  From a prebuilt thing or did you compile the sources?

Also, I see that we are on different versions of geos.  It's just a minor version and shouldn't make that big a difference... but you never know!

These are the things that I would check first, given that doing the same run as you but on my computer works.  By the way, here's what I ran:

```
└──> ~/private/pcb2gcode/pcb2gcode --mill-diameters=0.5 --bridgesnum=3 --isolation-width=0.7
└──> ~/private/pcb2gcode/pcb2gcode --mill-diameters=0.5 --bridgesnum=3 --isolation-width=0.6
```

And I did it in a directory that has the contents of your gerbers.zip in it, uncompressed.  It also has in there this this millproject file:

```
# general
metric=true
zero-start=true
metricoutput=true

# compatibility with GRBL
nog64=true
nog81=true

# milling
mill-diameters=0.2
zwork=-0.1
zsafe=1
mill-feed=200
isolation-width=0.8
mill-speed=10000

# drilling
zdrill=-1.9
drill-speed=10000
drill-feed=50
# G-code senders misinterpret G53, WTF
zchange-absolute=false
zchange=10
onedrill=true

### outline
cutter-diameter=0.8
zcut=-1.9
cut-feed=120
cut-vertfeed=20
cut-infeed=1
cut-speed=10000
# we are single-sided
cut-front=true
bridges=1
bridgesnum=4
zbridges=-0.8

# optimization
#path-finding-limit=50000
# workaround for https://github.com/pcb2gcode/pcb2gcode/issues/687
path-finding-limit=0
front=pcb-F_Cu.gbr
front-output=front.nc

outline=pcb-Edge_Cuts.gbr
outline-output=outline.nc
```

I think that it's pretty much the same as yours, yes?

Lycowolf, 2024-08-22:
Yes, it looks the same to me. The outputs you got is what I expected to get :).

----

I installed it using ArchLinux AUR package `pcb2gcode-git`. 

I had issues running the release version, because it depended on older version of libgeos; I don't remember the details, but I tried to just symlink it to the newer version and then abandoned it.

The git version was built IIRC the same day I reported this bug, using the master branch. The build process was just

```
autoreconf -i
  ./configure --prefix=/usr
  make -j

  make DESTDIR="$pkgdir/" install
```

in a correct directory (using a fakeroot or something like that).

I still have the symlink faking the geos version, but I just checked and it's not used: pcb2gcode is linked against `libgeos.so.3.12.2`:

```
pcb$  ldd /usr/bin/pcb2gcode                                                                                                                                                                                                          
        linux-vdso.so.1 (0x00007fff15d68000)
        libgerbv.so.1 => /usr/lib/libgerbv.so.1 (0x00007fbc266c8000)
        libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0 (0x00007fbc2657a000)
        libboost_program_options.so.1.83.0 => /usr/lib/libboost_program_options.so.1.83.0 (0x00007fbc26535000)
        libgeos.so.3.12.2 => /usr/lib/libgeos.so.3.12.2 (0x00007fbc26278000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007fbc25ff2000)
        libm.so.6 => /usr/lib/libm.so.6 (0x00007fbc25f03000)
...
```

eyal0, 2024-08-22:
So you did build pcb2gcode from sources...?

We'll, I'm willing to try whatever you suggest! If you can execute `git show` in the directory where you built pcb2gcode then I'll know what version you built from and I can make sure to build with that one and test again.

Also, I can upgrade my goes to the newer one that you are using and try that.

You and I are not on the same version of gerbv but that's very unlikely to be the problem!

I'm surprised that your pcb2gcode reports the git commit as "latest". I thought that it would have a commit SHA in there. Mine has the git commit listed in there. I forgot to put the version output from pcb2gcode into this bug report and I'm not home at the moment in order to do it.

Anyway, tell me what you want me to try and I'll do it. Otherwise, I'll probably first try the newer geos version while I wait for you to report to me the git commit SHA from which pcb2gcode was compiled, if you know it!

Lycowolf, 2024-08-23:
The commit I built pcb2gcode from is 2278a9ab.

I don't have the original cloned repo: I am using package building tools from my distro that don't keep the build directory and even do a new clone for each build. Nevertheless, the git hash is recorded in the package version ;-) (I didn't notice yesterday, because it was midnight in my TZ and I was quite tired.)

I would like to try building pcb2gcode using older libgeos in a container, but I don't know when I'll have the time: I hope I'll get to it this weekend.

eyal0, 2024-08-24:
That's the same version of code as mine.  I will try to upgrade geos and see if that makes a difference.  I see that you are also on a newer boost.  It might be that, too!  :sob: I will have to test both, I guess!  I think that I'm able to test those pretty easily.  Just hold tight and I'll figure it out.

For reference, here was yours, from above:

```
pcb2gcode --version                                                                                                                                                                                                         
2.5.0
Git commit: latest
Boost: 108300
Gerbv: 2.10.1-dev~ccf6a3
Geos: 3.12.2
```

Here is mine:
```
2.5.0
Git commit: v2.5.0-54-g2278a9ab
Boost: 106600
Gerbv: 2.9.8-dev~c559ea
Geos: 3.12.0dev
```

I doubt that gerbv is the issue because I use gerbv very little in pcb2gcode.  It's probably the geos or the boost.  I hope geos because boost is harder for me to work with!

eyal0, 2024-08-24:
Damn it the problem is boost.  I have rebuilt pcb2gcode and the version is now:

```
2.5.0
Git commit: v2.5.0-54-g2278a9ab
Boost: 108300
Gerbv: 2.9.8-dev~c559ea
Geos: 3.13.0dev
```

When I run it with isolation width 0.7, I get this:

[image]

We can see that both mine and yours now have the same problem.  :-(

I guess that I'll have to bisect boost!  Oy.

eyal0, 2024-08-25:
The bisect is quite complex.  I'll write down the process, as a note to myself, for next time.

```
pushd ~/private/boost/boost
git bisect #good or bad
git submodule update
rm -rf ../boost_from_src
./bootstrap.sh --with-libraries=program_options --prefix=${HOME}/private/boost/boost_from_src
./b2
./b2 install
popd
pushd ~/private/pcb2gcode
git clean -i -x -d
# answer yes to the above command
autoreconf -fvi
./configure BOOST_ROOT=/home/esoha/private/boost/boost_from_src pcb2gcode_CPPFLAGS_EXTRA="-Wall -Werror" --disable-silent-rules --enable-static --disable-shared --enable-code-coverage
make -j 20
popd
# Now test the code
pushd ~/Downloads/gerbers
~/private/pcb2gcode/pcb2gcode --mill-diameters=0.5 --bridgesnum=3 --isolation-width=0.7
image-diff-gui processed_front_final.svg processed_front_final-0.7.svg
popd
# If the image on the left has the wrong milling then it is "bad", otherwise, it is "good".
# back to the top to repeat the process.
```

eyal0, 2024-08-25:
Finally found the bad one:

```
└──> git show e54027791ae49cc0632851e52df7a0e2c86803a9
commit e54027791ae49cc0632851e52df7a0e2c86803a9 (HEAD, refs/bisect/bad)
Author: Automated Commit <automated@calamity.org.uk>
Date:   Wed Mar 2 11:00:02 2022 +0000

    Update geometry from master

Submodule libs/geometry e0d9c79c31..4a0af1b9e0:
  > Merge branch 'bg-prepare'
```

Obviously it's an update to a submodule.  The submodule in libs/geometry was updated from version e0d9c79c31 to 4a0af1b9e .  Okay, so now I get to bisect the submodule.  Hooray!  :sob: 

New procedure:

```
pushd ~/private/boost/boost
pushd libs/geometry
git bisect #good or bad
popd
rm -rf ../boost_from_src
./bootstrap.sh --with-libraries=program_options --prefix=${HOME}/private/boost/boost_from_src
./b2
./b2 install
popd
pushd ~/private/pcb2gcode
git clean -i -x -d
# answer 1 to the above command
autoreconf -fvi
./configure BOOST_ROOT=/home/esoha/private/boost/boost_from_src pcb2gcode_CPPFLAGS_EXTRA="-Wall -Werror" --disable-silent-rules --enable-static --disable-shared --enable-code-coverage
make -j 20
popd
# Now test the code
pushd ~/Downloads/gerbers
~/private/pcb2gcode/pcb2gcode --mill-diameters=0.5 --bridgesnum=3 --isolation-width=0.7
image-diff-gui processed_front_final.svg processed_front_final-0.7.svg
popd
# If the image on the left has the wrong milling then it is "bad", otherwise, it is "good".
# back to the top to repeat the process.
```

eyal0, 2024-08-25:
All done, here's the bad commit:

```
8774b7b2d7049ee9baf021204c772faa97de9642 is the first bad commit
commit 8774b7b2d7049ee9baf021204c772faa97de9642
Author: Barend Gehrels <barend@xs4all.nl>
Date:   Wed Feb 16 19:02:02 2022 +0100

    [major change] change the default, disable rescaling by default

 doc/release_notes.qbk                              |  8 +++++++
 include/boost/geometry/core/config.hpp             | 26 +++++++++++++---------
 .../geometry/strategy/spherical/envelope_range.hpp |  3 ---
 .../set_operations/union/union_other_types.cpp     |  2 --
 .../areal_areal/general_intersection_precision.cpp |  2 --
 .../overlay/areal_areal/interior_triangles.cpp     |  1 -
 .../overlay/areal_areal/intersection_pies.cpp      |  4 ----
 .../overlay/areal_areal/intersection_stars.cpp     |  4 ----
 test/robustness/overlay/areal_areal/intersects.cpp |  4 ----
 .../overlay/areal_areal/random_ellipses_stars.cpp  |  4 ----
 .../overlay/areal_areal/recursive_polygons.cpp     |  4 ----
 test/robustness/overlay/areal_areal/star_comb.cpp  |  4 ----
 .../overlay/buffer/recursive_polygons_buffer.cpp   |  4 ----
 13 files changed, 23 insertions(+), 47 deletions(-)
```

At least it's not too big.  I hope.

eyal0, 2024-08-25:
I spoke too soon.  This is a commit that is small but turns on a lot of code all over the place.

eyal0, 2024-08-25:
The code in boost geometry hints at some accuracy issue.  We have a method for that, using tolerance.  It makes the cuts very slightly less precise, like on the order of 0.0001 inches, but at least it will solve your problem.

Try to run your code with the tolerance parameter set at 0.0005 inches instead of 0.0004 inches.  Like this:

```
~/private/pcb2gcode/pcb2gcode --mill-diameters=0.5 --bridgesnum=3 --isolation-width=0.7 --metric=true --tolerance=0.0005
```

That ought to solve your problem.  It's probably too much for me to dig into boost and try to fix stuff there so, if you're satisfied with the result, I'll bump up the default tolerance to 0.0005 and we'll call it a day.

Try it and let me know if this solves your problem.  Thanks.

barendgehrels, 2024-08-26:
> 8774b7b2d7049ee9baf021204c772faa97de9642

Side note: It is always convenient to include a hyperlink to save some time searching through (very) old commits.

So you mean https://github.com/boostorg/geometry/commit/8774b7b2d7049ee9baf021204c772faa97de9642
of February 23, 2022, which changes our internal default from rescaling to non-rescaling in Boost 1.79, a very old version.

That was all as intended, and in the meantime the whole rescaling is removed in Boost.Geometry. 

So internally, we "rescaled" (which is similar but not identical to your fudge factor) - so it sounds logical that increasing the factor is related to removing the rescaling. But, without further info, I cannot say more about that.

For possible further questions or detailed cases, you might report an issue in https://github.com/boostorg/geometry/issues
including a minimal reproduction scenario, such that it can be investigated.

Cool you are using Boost.Geometry and I'm curious why for some cases you use Boost.Geometry, and for other cases you use geos, which is a library supporting the same functionality but implemented differently.

eyal0, 2024-08-27:
In practice, I found the geos was sometimes faster and more accurate.  But it was missing some features so I can't use it for everything.

Converting gerbv files to milling paths is quite complex!  There are a lot of geometry operations and even a Voronoi operation!  We get a lot of shapes that just barely overlap or meet at very shallow angles.  It's all one big kludge and it's kind of a miracle that it works at all!

I assume that boost just got better, not worse, and increasing the tolerance in my code is the right thing to do.  I don't know if it's worth making a minimal reproduction case because we'll probably just discover that boost is good and pcb2gcode is sloppy.  :-)

Lubensius, 2024-10-12:
I like to add some information here, as I'm experiencing the mentioned issue after updating to version 2.5.0. I was happily developing some new functionality under version 2.4.0 but had to pull and merge version 2.5.0 because of an Boost update to 1.74.0

$ pcb2gcode --version
2.5.0
Git commit: v2.5.0-58-g8c084afd-dirty
Boost: 107400
Gerbv: 2.8.1
Geos: 3.14.0dev

I tried all the above mentioned parameter changes and found also that I had not Geos installed on my development system. But after trying and installing it boiled down to the fact that only the "backtrack" parameter had a real consistent impact.
Setting backtrack=0 I got the issue no longer, but got very slow gcode generation, due to hundreds of senseless little fragments lifting the Z-axis.
Removing backtrack from millproject the issue reappeared.
By changing backtrack I could vary the number of fragments and the number of No-Sense milling path. With backtrack=1000 (metric) seems to be the point without issue for me.

BTW: the No-Sense paths cut the conductors on my layout
[image]

eyal0, 2024-10-14:
@Lubensius does slightly increasing the tolerance help you? The description of the solution is here: https://github.com/pcb2gcode/pcb2gcode/issues/688#issuecomment-2308709468

Lubensius, 2024-10-15:
@eyal0 I tried only shortly the effect of "tolerance" as I used it also in my software change (WIP). I have to decouple things first.
For what I saw it had an impact, but I will come back after further analyzing it.

tomjnixon, 2025-12-28:
#743 fixes a lot of issues for me, but it's still possible for pcb2gcode to trigger issues in boost geometry (at least when not built with geos).

The example in https://github.com/boostorg/geometry/issues/1428#issuecomment-3694915140 comes from this project, with the current master ( 0029d61225fb3f3bd3ac8b4e21ba21447bad9c45 ) and boost 1.90, though all versions after 1.77 have similar problems:

[bg_fail_example.zip](https://github.com/user-attachments/files/24361280/bg_fail_example.zip)

Here, this call to buffer returns empty or nonsense geometry, so the `PathFindingSurface` is missing a keep-out zone: https://github.com/pcb2gcode/pcb2gcode/blob/0029d61225fb3f3bd3ac8b4e21ba21447bad9c45/surface_vectorial.cpp#L985

(i don't have permission to re-open this issue, but it might be helpful while something is still broken)

eyal0, 2025-12-28:
Okay, I have reopened. I continue to be on vacation and won't loll until January.

Boost is not as good as Geos, for as much as I want boost to be perfect!

All the buffering goes through a single function. Sometimes I can find tricks to convince boost to do the right thing!

To start, I need to find a geometry that is a legal input to buffer but the output is illegal. Then I can try to categorize the issue and find a general solution.

I'll look in January. Thanks!

tomjnixon, 2025-12-28:
> Okay, I have reopened. I continue to be on vacation and won't loll until January.

No rush at all!

> To start, I need to find a geometry that is a legal input to buffer but the output is illegal. Then I can try to categorize the issue and find a general solution.

This code contains such a geometry, hopefully that helps: https://github.com/boostorg/geometry/issues/1428#issuecomment-3694915140