pcb2gcode/pcb2gcode #691: Request: Add option to use G1 commands in milldrill

calischs, 2024-08-26:
Thanks so much for writing this tool -- it's been very helpful so far!

## What did you try (include command-line arguments):

I'm trying to use pcb2gcode with my fiber laser to make PCBs.  My workflow is EasyEDA > pcb2gcode > Lightburn.

I am using milldrill to create drill gcode (i.e. `min-milldrill-hole-diameter=0` in my millproject file).

## What happened:

The gcode I created for front and outline is imported flawlessly into Lightburn.  When I try to import drill gcode, Lightburn says "Nothing imported".

## What did you expect to happen:

I expected the drill gcode file to import like the others.  I believe this happened because Lightburn cannot handle G2/G3 arc commands.

Is it possible to add an option to use G1 commands instead of G2/G3 when milldrilling?  I know this would create plunges instead of helices, which isn't the best for milling, but for laser cutting this is no problem.

Being able to use milldrilling via Lightburn would be an extremely useful capability for me!
 

## pcb2gcode version (run pcb2gcode --version to see this):

2.5.0

eyal0, 2024-08-26:
Let's find a solution for you!

First of all, converting G2/G3 to G1 would not make plunges, it would just convert the arcs into tiny sections that approximate an arc.  Do you agree?  The G2 and G3 commands draw an arc in 3 dimensions, a helix, like you noticed.  Is your goal to convert the helix steps into a bunch of short G1 steps instead?  Is the laser actually able to mill a hole through the PCB?  That's cool!

The milldrill option is generally for people that don't want to drill.  They use the milldrill operation so that the milling bit will cut away at the PCB, little by little, to make a hole for them.  For people that actually have drilling capability, they should probably just use regular drilling and drill a hole.  I would assume that a laser is more like a drill because it can, like, burn a hole through the PCB?  I don't really know how it works.

Anyway, let me know more about how you want pcb2gcode to work and we'll come up with a solution.  If really all you need is to convert G2/G3 to G1s, there are probably tools that already exist to do that!  Or if not, we could make one.  I'm usually hesitant to add more and more features to pcb2gcode if they can be external tools because pcb2gcode is becoming unwieldy with all the stuff that's already built into it!

calischs, 2024-08-26:
Thanks for responding so quickly!

Here's a great example of making a fine pitch pcb with laser: https://www.kurokesu.com/main/2021/01/07/making-fine-pitch-pcb-prototypes-with-fiber-laser/
And here's a video using the fiber laser for drills: https://youtu.be/PoYcjyghDx4?si=CnGkmL0hYlRRHVTY&t=871.
All the examples I've seen online either require a lot of manual processing of the files (e.g. with flatcam, which I can't get to run due to a Qt4 dependency) or are a limited in the lasers they work with / don't give any previewing (e.g., with Balor https://www.bryce.pw/engraver.html).  Don't get me wrong, both of those tools are great -- I'm just trying to find a toolchain that meets my needs.  I think using pcb2gcode together with Lightburn might be the best of both worlds?

Anyway, responding to your questions:
1. Agreed on plunges.   My goal is to have gcode without any G2/G3 commands in it, so yes, converting the helix steps to a bunch of G1 steps would likely work. (There's a small chance lightburn won't like G1s with a Z component, but I'm guessing it'll be fine).
2. I'd like to use the laser in milldrill operation so I can make different sized holes for the different components on my PCBs.  You can think of the laser like an endmill with a very small but fixed diameter. FWIW, the project I'm working on currently only has circular holes, but I can imagine many times when noncircular holes are also useful. 
3. I did look around for external tools to convert G2/G3 to G1 -- I found one javascript-based tool (https://constant-flow.github.io/arc-breaker/) but it produced bugs for me and it's a pain to send every file to a website.  I'd love to have a CLI tool so I can iterate quickly.  Ideally it would be packaged as part of pcb2gcode, but I could imagine running a post-processing command for the drill file afterwards.  I hear you on not bloating pcb2gcode. :)  FWIW, I have run routers/mills that do not like G2/G3, so this feature would likely have benefits beyond just lasers.

eyal0, 2024-08-27:
Alright, it makes sense to me.  I think that the arc-breaker tool looks promising.  Yes, it's a website, but it's also just some javascript code.  Converting that code to python would be easy.  Or, just run it as javascript using nodejs.  Do either of those alternatives suit you?  You're already doing a few tools so this would just be one more, not too bad.

Would nodejs be okay with you?  And then we could leave it as javascript and not rewrite it.  Or do you prefer python or something else?  And then we need to rewrite it.

The code is here: https://github.com/constant-flow/arc-breaker/blob/main/index.html#L211-L382

From the comments in the code, looks like it is based on the Marlin code, which is probably going to be well-written!

calischs, 2024-08-27:
So far I haven't gotten arc breaker to give me bug free results.  Below is an example inputing the milldrill file from pcb2gcode with a bunch of NaNs in the output.  It looks to me like the code isn't expecting 3 axis G1 moves -- maybe that's why it barfs?  Screenshot and zip attached.

[image]

[arcbreaker-example.zip](https://github.com/user-attachments/files/16755992/arcbreaker-example.zip)

eyal0, 2024-09-13:
I don't imagine that I'll have time enough to fix that code.  Are you able?

Fanjio01, 2024-09-29:
Hi,
In case it helps, this tool works fine to convert G2/G3 to G1 in pcb2gcode's files: https://github.com/FormerLurker/ArcWelderLib

It's a command line tool, pre-compiled for multiple platforms including Linux and Windows, containing a binary that does the conversion job:
./ArcStraightener {pcb2gcodeSourceFile.cnc} {targetFile.cnc}

Thanks

eyal0, 2024-09-29:
I knew about arcwelder but I thought that it could only create arcs, not also dismantle them.  That would be a good solution for this issue!

Fanjio01, 2024-09-30:
I don't have much knowledge about this tool,, but as far as i've seen there are 2 binaries provided: one to create and one to dismantle arcs.
I've used the dismantle binary successfully yesterday on a milldrill pcb2gcode output file to convert G2/G3 to G1.
Regards