pcb2gcode/pcb2gcode #698: Feature: Add support for reprapfirmware

timaydin, 2025-03-10:
I am using pcb2gcode with a mini CNC that has a Duet3 6HC card running reprapfirmware. The GCODE dialect of this firmware is different in a few areas when compared to what Mach3, Mach4, or LinuxCNC supports. I have  forked pcb2gcode and made modifications so that reprapfirmware is supported, but the modifications are not generic. In order to add reprapfirmware support, the heading in the GCODE output needs to be configured based on what GCODE dialect is active. For example, the standard GCODE output uses

S <speed>

to set the spindle RPM. This does not work with reprapfirmware. There are also other gcodes that behave slightly different. Also, reprapfirmware does not support variable assignments in the following form

#xxxx = #yyyy

so I cannot use the built in auto levelling. Instead, I have to use reprapfirmware's builtin autolevelling. And in order to do that, I have to use the M557 command as give it the board outline size so that the probing can be done in that area only.

So my question is, what would be the best strategy to add support for reprapfirmware so that it can be merged into the mainline? I am willing to work on this.

eyal0, 2025-03-11:
pcb2gcode already has so many flags that I worry about adding new ones!  If this is something that can be done with a script, post-processing the output, that would be great!  But if you feel that it would be too difficult then maybe a flag is needed.

There are already many attempts to support various forms of gcode and I get lots of them.  Often for various gcode of different CNC machines, sometimes gcode for Marlin-based machines, and even gcode for laser etching.

If you think that it can't be done with a script and you really want a flag, okay.  Go ahead and work on the code and submit a patch and I'll look at it.  The CI, unfortunately, broke recently so I'm wary of adding changes that can't get through CI.  If you want to fix CI, that would help, too!  I tried but I didn't have the time to work on it.  But I'll look at whatever PR you can generate.

timaydin, 2025-03-11:
Here is the summary of the changes I want to make:

1) Add new general option, like --controller. The values can be Mach3, Mach4, LinuxCNC, and RepRapFirmware and others. The --software option can either stay and only be useful for pcb2gcode's own autolevelling support, or it can be made an alias to the new --controller option, and pcb2gcode's own auto levelling support can be tied to the new option.

2) For gcodes that are controller specific, add functions that implement the action based on the gcode option. For example, gcode_set_spindle_rpm will use the appropriate gcode to set the spindle rpm. Similarly, gcode_controller_auto_levelling implements the controllers own auto levelling support. Any other gcode that is controller specific will be wrapped in its own gcode_xxxx function.

Let me know what you think about this approach. If you agree, I will convert my existing modifications accordingly and come up with a pull request.

eyal0, 2025-03-12:
You'd be surprised at how weird the controllers are.  In an ideal world, we would just be able to make a name for each controller type and be done with it.  But in practice, we've got people doing all sorts of stuff like etching their board with a laser but the laser uses Marlin firmware.  Or drawing a gerber layer on to a PCB using a pen attached to a 3d printer.  There's a lot of stuff going on which is why there are all these commands like "nom6" and such.  I don't know how easy it will be to group them into controllers.

I'm not as involved with this project as I used to be so if you can come up with something good, I'll merge it.  Even if you need to change the existing flags and it breaks some old stuff, that seems okay with me if the result is better than the original.