pcb2gcode/pcb2gcode #700: Cannot compile on Fedora 41 - autoconf errors.

RWL-69, 2025-03-14:
## Description of bug:
Downloaded latest version to test.  Installed all the tools and tried to run the commands as in the README.md.  

First error was ./configure wouldn't run.  Needed to run autoconf first which lead to errors.

## What did you try (include command-line arguments):
1. First step - ./configure
2. Second step - autoconf
3. More steps...  autoupdate as suggested by autoconf

## What happened:
autoconf
configure.ac:77: warning: The macro 'AC_LANG_CPLUSPLUS' is obsolete.
configure.ac:77: You should run autoupdate.
./lib/autoconf/c.m4:270: AC_LANG_CPLUSPLUS is expanded from...
configure.ac:77: the top level
configure.ac:120: warning: The macro 'AC_HEADER_STDC' is obsolete.
configure.ac:120: You should run autoupdate.
./lib/autoconf/headers.m4:663: AC_HEADER_STDC is expanded from...
configure.ac:120: the top level
configure.ac:129: warning: AC_OUTPUT should be used without arguments.
configure.ac:129: You should run autoupdate.
configure.ac:14: error: possibly undefined macro: AM_SILENT_RULES
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:16: error: possibly undefined macro: AM_INIT_AUTOMAKE
configure.ac:21: error: possibly undefined macro: AC_PROG_LIBTOOL
configure.ac:62: error: possibly undefined macro: AC_DEFINE

autoupdate 
configure.ac:120: warning: The preprocessor macro 'STDC_HEADERS' is obsolete.
  Except in unusual embedded environments, you can safely include all
  C89 headers unconditionally.

 autoconf
configure.ac:117: warning: The preprocessor macro 'STDC_HEADERS' is obsolete.
configure.ac:117:   Except in unusual embedded environments, you can safely include all
configure.ac:117:   C89 headers unconditionally.
configure.ac:11: error: possibly undefined macro: AM_SILENT_RULES
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:13: error: possibly undefined macro: AM_INIT_AUTOMAKE
configure.ac:18: error: possibly undefined macro: AC_PROG_LIBTOOL
configure.ac:59: error: possibly undefined macro: AC_DEFINE

Trying to run ./configure gives this error.

./configure: line 2578: syntax error near unexpected token `yes'
./configure: line 2578: `AM_SILENT_RULES(yes)'

## What did you expect to happen:

Compile the program.

## pcb2gcode version (run pcb2gcode --version to see this):
pcb2gcode-latest.tar.gz  downloaded 2025-03-14
Same issues in 
pcb2gcode-2.5.0.tar.gz

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**  You might need to zip them.  If you don't provide a millproject file, make sure that you copy-paste the command-line that you used.

eyal0, 2025-03-14:
Did running autoconf -fvi solve your issue?

I'm actually not a pro at autoconf. I just run that command and it seems to work for me. Or maybe autoreconf? I forget.

RWL-69, 2025-03-15:
I am not a programmer, just enough knowledge to be trouble.  :)

Tried autoconf -fvi, tried autoreconf -fvi still have errors.

After running both.

autoconf 
configure.ac:108: warning: The macro 'AC_TRY_COMPILE' is obsolete.
configure.ac:108: You should run autoupdate.
./lib/autoconf/general.m4:2845: AC_TRY_COMPILE is expanded from...
lib/m4sugar/m4sh.m4:690: _AS_IF_ELSE is expanded from...
lib/m4sugar/m4sh.m4:697: AS_IF is expanded from...
./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
./lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
m4/ax_c_compile_value.m4:91: AX_C_COMPILE_VALUE is expanded from...
configure.ac:108: the top level
configure.ac:117: warning: The preprocessor macro 'STDC_HEADERS' is obsolete.
configure.ac:117:   Except in unusual embedded environments, you can safely include all
configure.ac:117:   C89 headers unconditionally.

The main point is STDC_HEADERS is obsolete.

From 
https://www.gnu.org/software/autoconf/manual/autoconf-2.69/html_node/Obsolete-Macros.html

— Macro: AC_STDC_HEADERS

    Replaced by AC_HEADER_STDC (see [AC_HEADER_STDC](https://www.gnu.org/software/autoconf/manual/autoconf-2.69/html_node/AC_005fHEADER_005fSTDC.html#AC_005fHEADER_005fSTDC)).

so a change that needs to be made to the config file.

RWL-69, 2025-03-19:
Looking further into this issue.  Found an issue in m4/ax_c_compile_value.m4 with an obsolete macro.  
Reported the issue on autoconf site.

Changed ax_c_compile_value.m4 with the help of Copilot and was able to compile program.  This was the change I made that worked.  Delete the code, including AC_DEFUN and replace.  

```

AC_DEFUN([AX_C_COMPILE_VALUE], [
  pushdef([ac_c_compile_value],
    AC_TR_SH_REUSE([ac_cv_c_compile_value_$2]))dnl
  ac_c_compile_value_expand="$1"
  AC_CACHE_CHECK([value of $1 by analyzing object code],
                 ac_c_compile_value, [
    save_CFLAGS="$CFLAGS"
    CFLAGS="$CFLAGS -c -o conftest.o"
    AC_COMPILE_IFELSE([[$3
      #include <stddef.h>
      #include <stdint.h>
      #include <stdlib.h>
      #define COMPILE_VALUE $ac_c_compile_value_expand
      #define HEX_DIGIT(n)      ((n) >= 10 ? 'a' + (n) - 10 : '0' + (n))
      char object_code_block[] = {
        '\n', 'e', '4', 'V', 'A',
        '0', 'x',
        (char) HEX_DIGIT((((COMPILE_VALUE / 16) / 16) / 16) % 16),
        (char) HEX_DIGIT(((COMPILE_VALUE / 16) / 16) % 16),
        (char) HEX_DIGIT((COMPILE_VALUE / 16) % 16),
        (char) HEX_DIGIT(COMPILE_VALUE % 16),
        'Y', '3', 'p', 'M', '\n'
      };]],
      [ac_c_compile_value=`
        typeset -i n=\`sed -ne 's/^e4VA0x\(.*\)Y3pM$/0x\1/p' < conftest.o\`;
        echo $n`],
      [ac_c_compile_value=0])
    CFLAGS="$save_CFLAGS"])
  AC_DEFINE_UNQUOTED(AC_TR_CPP_REUSE(compile_value_$2),
                     [$[]ac_c_compile_value],
                     [$1])
  popdef([ac_c_compile_value])dnl
])

```

Now to test.

closed 2025-03-29 by RWL-69

RWL-69, 2025-04-01:
Just adding that so far it is working as expected.  No errors in my testing yet.