pcb2gcode/pcb2gcode #701: Mill cuts into trace

Eerolz, 2025-03-21:
## Mill (sometimes) cuts into trace when doing 2 right angle turns
I made a small test Gerber that I'm testing different settings with, and noticed pcb2gcode cuts into the trace at a spot with 2 right angle turns after one another.

## What did you try (include command-line arguments):
1. Setup millproject file 
2. `pcb2gcode --front test.GBR`
3. Tried with different offsets, happened from offsets `0.06<offset<=0.29`
## What happened:
I got the clearance requirement warning, and when inspecting the SVG, I saw tool path lines on one of the corners extending all the way to the trace.
## What did you expect to happen:
The toolpath drawn successfully without cutting into the traces.
## pcb2gcode version (run pcb2gcode --version to see this):
```
2.5.0
Git commit: latest
Boost: 108300
Gerbv: 2.10.0-1build2
Geos: Not installed
```

[test_issue.zip](https://github.com/user-attachments/files/19392429/test_issue.zip)

eyal0, 2025-03-22:
Previously someone complained about something similar and adjusting the tolerance setting helped. Can you try that? Make tolerance just a bit bigger.

Eerolz, 2025-03-22:
Thanks for reply! 
I tried playing around with tolerance, but only thing it did was make the warning about it happening disappear. 
I also realised its a whole loop it makes, not just running into the edge and back. Decided to attach all the files just in case it helps.

[test-files.zip](https://github.com/user-attachments/files/19404098/test-files.zip)

eyal0, 2025-03-22:
The other thing that you can luck with is the path finding optimizations. It makes mistakes sometimes due to floating point math that is difficult. Try setting the path finding limit down to zero.

The options are described in the wiki here on GitHub.

Eerolz, 2025-03-22:
I now tried the options in optimisations, couldn't find one that'd prevent this from happening, but at least if I do `optimise=0` I get a point in the gcode in the intersection, so its just quickly removing 3 lines of gcode. 

Yeah, I checked out the wiki and have now tested/tried most if not all options in General, Milling and Optimizations. Manually fixing the mistake will probably be the easiest solution for now.

eyal0, 2025-03-23:
I am currently on vacation so I can't look into it further.

The optimization flag uses a path algorithm to decrease the accuracy of paths in exchange for making the file smaller. Setting that flag to zero will make your file larger but otherwise should have no negative effect.

eyal0, 2025-04-02:
Here's what it looks like on my computer:

[image]

I don't see any traces getting cut into.  I am using pcb2gcode with geos:

```
2.5.0
Git commit: v2.5.0-59-gbb2f1478
Boost: 106600
Gerbv: 2.9.8-dev~c559ea
Geos: 3.13.0dev
```

The geos geometry library seems faster and more accurate than the boost one, which is what you will use by default if you don't have geos install.  Your easiest solution might be to install geos.  It's not difficult, you probably have it on your Linux system available.  Maybe you type `sudo apt-get install libgeos++-dev` to install it.  And then you run `autoreconf -fvi` again and it will find the geos and use it.

Let me know how that goes.  If it solves your problem, perhaps close this issue.

Eerolz, 2025-04-07:
Sorry for taking a bit to respond.

Yes, looks like geos fixed the issue, I had tried figuring out how to install it before but had not done the `autoreconf -fvi`. 

Thanks for helping me solve this issue, and hope you have a great day.

closed 2025-04-07 by Eerolz