pcb2gcode/pcb2gcode #702: Question - Has --dpi option been depreciated?

RWL-69, 2025-03-31:
Hello,  Finally got pcb2gcode to compile and I am trying to setup for my laser etcher.  I tried to set the dpi using options in the millproject file and by command line.

In both cases I get 

> pcb2gcode --dpi=4000 --back Photodiode_amp-B_Cu.gbr 
Error: You've supplied an invalid parameter.
Details: unrecognised option '--dpi=4000'

Or

> pcb2gcode --back Photodiode_amp-B_Cu.gbr
Error parsing configuration file "millproject": unrecognised option 'dpi'

If it has been, the Manual needs to be updated to remove this option.  
Thank you

RWL-69, 2025-04-01:
Is there another wiki than these?    DPI is still listed here.   All searches for pcb2gcode and wiki, lead me to these two main links and others that don't mention being obsolete or any other wiki.  Even the wiki link on the top of the project page leads to one of these links.

[https://github.com/pcb2gcode/pcb2gcode/wiki/Manual](url)

[https://github-wiki-see.page/m/pcb2gcode/pcb2gcode/wiki/Manual](url)

eyal0, 2025-04-01:
I meant these: 

[image]

RWL-69, 2025-04-01:
Okay.  I don't know why the wiki isn't clear about this.   

I finally see that is happening.  There are multiple github wiki's for pcb2gcode.  I have ended up on the ones with full descriptions and options for the landing page.  They should be removed.  This was the page I was seeing and bookmarked.  No mention of being obsolete.  I also found a couple of clone sites from the below with no mention of being obsolete.

[image]

[pcb2gcode manual page](https://github.com/pcb2gcode/pcb2gcode/wiki/Manual)

Thanks for clarifying this.

eyal0, 2025-04-01:
The beauty of a wiki is that anyone can go in an edit it.  Be my guest!

RWL-69, 2025-04-01:
In my line of work, permission is necessary.  As I learn, I will make changes then.  Thank you.

closed 2026-11-14 by eyal0