pcb2gcode/pcb2gcode #705: Which C++ skill/knowledge needed for contributing at this project

R4id0ss, 2025-04-30:
Hello, I wanted to thank you for all the work done. I use your software in my daily life.
That said, I would like to start contributing with some ideas but I do not have complex knowledge of C++.

I would like to know from you what skills on C++ are necessary to be able to contribute concretely also having ideas but without weighing on your free time. 

THANK YOU SO MUCH for the work done.

eyal0, 2025-05-01:
Yeah, there's a lot of c++ there!  And much of it is quite complicated.

If you're looking to contribute, there are plenty of open issues and pull requests than need attention.  I haven't been able to get to all of them.

Is there something specific that you'd like to improve?

R4id0ss, 2025-05-01:
> Is there something specific that you'd like to improve?

one of the things that i think needs to be reworked is the double sided pcb alignment. In my opinion, the possibility that the edges are not parallel is not considered and this leads, when you turn the pcb, to have an alignment hole that is not able to tell you if the board has undergone a "rotation". if for example you were to insert into the code the possibility of inserting "front coordinates" which are the alignment points (at least 2) taken at the beginning and then on the basis of the alignment points taken after the rotation generate the back with the necessary corrections we would create a front-back alignment that has an error under 100 micrometers.

eyal0, 2025-05-05:
If you can imagine how this might work, we can see what can be done.

Currently, you can specify a mirroring axis, either x or y, and you can specify a offset.  If you put a couple of holes along that axis, you can use those to do the mirrors by putting studs into the CNC plate.

I think that someone proposed allowing the user to specify which holes to use for mirroring.  But what does that even mean "which holes"?  Holes don't have names.

I think that the first step would be to imagine how it might work and then let's see if it can be done.

R4id0ss, 2025-05-10:
I've found a project which the same idea but made with scala.

So, for now, I'll concentrate on other issues.

closed 2026-11-13 by eyal0