pcb2gcode/pcb2gcode #706: millingn repeatedly bumping up and down in the same spot [video provided]

thomergil, 2025-05-01:
My millproject:

```
# generic
metric=true
metricoutput=true

# tracing
zwork=-0.06          # Depth in mm
zsafe=2              # Height for movements
zchange=35           # Height for tool changes
mill-feed=100        # Feed rate in mm/minute
mill-speed=12000     # RPM
nom6=1               # Don't issue M6 command
spinup-time=3.0      # Time to spin up the spindle in seconds
spindown-time=3.0    # Time to spin up the spindle in seconds
backtrack=0          # Don't cross the copper; always pull up first
isolation-width=0.58 # As thick isolation as possible without running into trouble
milling-overlap=75%  # Re-mill with 75% overlap as part of creating isolation width

# Voronoi regions
voronoi=1

# drilling
zdrill=-1.7
drill-feed=60
drill-speed=10000
nog81=1              # Use G0/G1 instead of G81/G80
onedrill=1           # Don't swap drills

# this will mess up offset! don't use!
# zero-start=true
```

My command:

```
pcb2gcode --back decibel-meter-B_Cu.gbr --mill-diameters 0.169 --x-offset -70
```

The result is fine, but the process gets repeatedly stuck bumping up and down in the same spot. Here's a video of what is happening:

https://vimeo.com/1080537175/23c2a2fe85?share=copy

eyal0, 2025-05-05:
Without including the gbr files, I'm unable to try this myself.  But if you can provide the files, I can run the code and then use `bCNC` to examine the generated CNC output and see what's going on.  Maybe it's trying to mill a really tiny area and it's unable to make it a single pass?  I dunno.  Please send the .gbr files.

If you want to keep your design private, you can also try triming the .gbr files to include just the part of the design that is doing that behavior.  You can use `gerbv` to trim the .gbr file.

Also, here's the version of pcb2gcode that I use:

```
└──> ~/private/pcb2gcode/pcb2gcode --version
2.5.0
Git commit: v2.5.0-59-gbb2f1478
Boost: 106600
Gerbv: 2.9.8-dev~c559ea
Geos: 3.13.0dev
```

thomergil, 2025-05-05:
[decibel-meter-B_Cu.gbr.txt](https://github.com/user-attachments/files/20042906/decibel-meter-B_Cu.gbr.txt)

eyal0, 2025-05-05:
That milling path is intense!

[image]

You're asking for isolation of 0.58 but your milling tool is 0.169 in width and you want 75% overlap.  That means that you will have 0.169 cleared on the first pass and then 0.25% of 0.169 cleared on each additional pass, to reach 0.58.  So the number of additional passes will be x in `0.169 + x*0.25*0.169 = 0.58`.  That's 9.7.  So rounded up, that's 10.  Indeed, I see that there are 10 additional passes in the image above.  So 11 passes total.  Okay!  If that was your intention, I'm not going to question it!

The little tiny movements that you see are due to tiny "spikes", which is a feature that I added a while ago.  I'll explain: The spikes are there to deal with failures of overlap.  In the image here: https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Milling#--milling-overlap

[image]

Imagine that you didn't have that spike in the milling.  What would happen is that the bit wouldn't properly tuck into that corner there and you'd have some material that wasn't remove.  The "spike" gets it into the corner so that, even without a milling overlap, it'll still work.

One thing that pcb2gcode can do is recognize that the spikes are connected to the rest of the milling.  So what it does is it can draw the spike and then scooch back, essentially milling over the spike twice.  However, if you have turned off backtracking, it can't do that.  Turning on backtracking makes a path that is much smaller.  bCNC says that it'll mill about twice as fast.  I also noticed that pcb2gcode ran WAY faster.

We can see the difference from enabling backtracking here:

[image]

See how instead of making a bunch of disconnected rings, it connected the rings?  That allows it to not have to rise up between each ring, which saves a bunch of time.  There is extra milling to connect those passes but its just going over material that you wanted removed anyway.

Want to try the backtracking feature, set to 1, and see if it helps?  Also, overlap of 25% or even 10% ought to be enough.  Spikes take care of all the rest.  flatcam didn't have spikes so if you're coming from flatcam, this may be where you got the idea that 75% overlap is needed?  I dunno, I don't know much about flatcam.

I hope that this helps you!

thomergil, 2025-05-05:
Wow, cool, thank you. I will take a more precise look at your comments and suggestions. But I did not come from flatcam. My path is KiCad => pcb2gcode => OpenCNCPilot.

thomergil, 2025-05-05:
...and I don't know what I'm doing! This is my first adventure with any of this. :)

eyal0, 2025-05-05:
Try an overlap of 25% and backtracking equal to 1 and see if you like those results. I use bCNC to view a path before I run it. pcb2gcode will generate SVG files that can show you the paths as they are created.

thomergil, 2025-05-09:
This works great. I've described my process and use of `pcb2gcode` here: https://github.com/thomergil/pcb-nomad3