pcb2gcode/pcb2gcode #707: UI is poor (arguments checked one-by-one, cryptic error messages when arguments have problems)

michaeljtbrooks, 2025-05-04:
## Description of bug:
The UI is poor:

a) Stops parsing at first argument error it finds. Leading the user to play whac-a-mole, fixing each argument problem one-by-one rerunning the command each time. It should parse all arguments and pull out all issues (as much as possible, I'm aware some arguments going to be conditional on what has been entered for others), and should tell the user about all the problems found all at once.

b) Errors don't tell a user how to amend the situation.
e.g. "Error: Tool changing height not specified."
Would be far better as:
"Tool changing height not specified. Specify --zchange=<integer> to set the tool change height, e.g. --zchange=25"

## What did you try (include command-line arguments):
1. First step
`pcb2gcode --front=AVmic3-F_Cu.gbr --drill=AVmic3.drl \
    --isolation-width=0.3 --mill-diameters=0.3 --cut-infeed=-0.1 \
    --mill-feed=100 --mill-speed=15000 --safety=3 \
    --drill-feed=50 --drill-speed=10000 --drill-diameters=0.8 \
    --output-dir=./front_gcode --basename=AVmic3_front
`

2. Second step
`
pcb2gcode --metric --metricoutput \
    --front=AVmic3-F_Cu.gbr --drill=AVmic3.drl \
    --mill-diameters=0.3 --zwork=-0.1 --zsafe=20 \
    --mill-feed=100 --mill-speed=15000 \
    --drill-feed=50 --drill-speed=10000 --zdrill=-2 \
    --drill-side=front \
    --output-dir=./front_gcode --basename=AVmic3_front
`

3. Third step
`
pcb2gcode --metric --metricoutput \
    --front=AVmic3-F_Cu.gbr --drill=AVmic3.drl \
    --mill-diameters=0.3 --zwork=-0.1 --zsafe=20 \
    --mill-feed=100 --mill-speed=15000 \
    --drill-feed=50 --drill-speed=10000 --zdrill=-2 \
    --drill-side=front --zchange=25 \
    --output-dir=./front_gcode --basename=AVmic3_front
`

4. Fourth step:
`
pcb2gcode --metric --metricoutput \
    --front=AVmic3-F_Cu.gbr --drill=AVmic3.drl \
    --mill-diameters=0.3 --zwork=-0.1 --zsafe=20 \
    --mill-feed=100 --mill-speed=15000 \
    --drill-feed=50 --drill-speed=10000 --zdrill=-2 \
    --drill-side=front --zchange=25 \
    --basename=AVmic3_front
`

## What happened:

Specified this:
`pcb2gcode --front=AVmic3-F_Cu.gbr --drill=AVmic3.drl \
    --isolation-width=0.3 --mill-diameters=0.3 --cut-infeed=-0.1 \
    --mill-feed=100 --mill-speed=15000 --safety=3 \
    --drill-feed=50 --drill-speed=10000 --drill-diameters=0.8 \
    --output-dir=./front_gcode --basename=AVmic3_front
`

Got this:
`
Error: You've supplied an invalid parameter.
Details: unrecognised option '--safety=3'
`

Cross-checked it against the --help. Realised the correct argument was --zsafe. 

Tried this:
`
pcb2gcode --metric --metricoutput \
    --front=AVmic3-F_Cu.gbr --drill=AVmic3.drl \
    --mill-diameters=0.3 --zwork=-0.1 --zsafe=20 \
    --mill-feed=100 --mill-speed=15000 \
    --drill-feed=50 --drill-speed=10000 --zdrill=-2 \
    --drill-side=front \
    --output-dir=./front_gcode --basename=AVmic3_front
`

Got this:
`
Error: Tool changing height not specified.
`

Had to consult --help again to find the command for tool changing height.

Finally tried this:
`
pcb2gcode --metric --metricoutput \
    --front=AVmic3-F_Cu.gbr --drill=AVmic3.drl \
    --mill-diameters=0.3 --zwork=-0.1 --zsafe=20 \
    --mill-feed=100 --mill-speed=15000 \
    --drill-feed=50 --drill-speed=10000 --zdrill=-2 \
    --drill-side=front --zchange=25 \
    --output-dir=./front_gcode --basename=AVmic3_front
`

and it appeared to work, but then no files got dropped anywhere on the filesystem.

Removing the --output-dir finally gave me some files output in the current directory.

`
pcb2gcode --metric --metricoutput \
    --front=AVmic3-F_Cu.gbr --drill=AVmic3.drl \
    --mill-diameters=0.3 --zwork=-0.1 --zsafe=20 \
    --mill-feed=100 --mill-speed=15000 \
    --drill-feed=50 --drill-speed=10000 --zdrill=-2 \
    --drill-side=front --zchange=25 \
    --basename=AVmic3_front
`

## What did you expect to happen:

All arguments, insofar as possible, are checked for errors and all errors found are shown to the user at the same time all in one go. The user can then fix the command in one attempt.

Error messages shown to users should include suggestions for how to rectify the situation where possible.

## pcb2gcode version 2.5.0

(This is a UI issue, not a file output issue.)

eyal0, 2025-05-07:
This is probably an easy thing to do but its a large modification.  Is this something that you would like to do?  Maybe the user from #705 would like to do this?

In all places where an error in configuration is noticed, add to a string that is collecting all the errors and then, if there were any errors, display them all and quit.

BTW, as a workaround, there is an option to treat configuration errors as warnings.  Could you check if this solves your problem partially?

R4id0ss, 2025-05-10:
> All arguments, insofar as possible, are checked for errors and all errors found are shown to the user at the same time all in one go. The user can then fix the command in one attempt.
> 
> Error messages shown to users should include suggestions for how to rectify the situation where possible.
>

Hi,
It's a good idea.

I would like to implement this enhancement.

I'm still reading and understanding the actual code. 

I've added to my to do list.

closed 2026-11-14 by eyal0