pcb2gcode/pcb2gcode #714: Milldrill output missing G17 for arc commands, causing errors on CNC controllers

thomergil, 2025-05-31:
The G-code generated by `pcb2gcode` for `milldrill` operations relies on arc commands (e.g., G02, G03) for helical drilling. However, the output does not consistently include a `G17` command to select the XY plane, which is typically required for these arc commands to function correctly.

When G-code generated by `pcb2gcode` (specifically for `milldrill` operations) that lacks a `G17` is run on a CNC controller (e.g., OpenCNCPilot), it results in errors stating that arc commands require a plane to be selected. The machine will not execute the G-code.

thomergil, 2025-05-31:
I am closing this bug. It appears the workaround is NOT the `G17` but performing a Homing and GRBL reset prior to running the milldrill G-code. There is some funny soft state that lingers that affects G2/G3 arc commands.

eyal0, 2025-05-31:
Good to know. And now it's recorded here so someone can search for it!