pcb2gcode/pcb2gcode #719: Processing input files... Internal error in gerberimporter...

Terak8510a, 2025-06-18:
## Description of bug:
I think the importer did not like something in one of the layers.

## What did you try (include command-line arguments):
pcb2gcode --back=/home/user/CAD/KiCad/PROJECTS/nano-v3/NanoV3.3/attempt-1/NanoV3.3-B_Cu.gbr --drill=/home/user/CAD/KiCad/PROJECTS/nano-v3/NanoV3.3/attempt-1/NanoV3.3.drl --front=/home/user/CAD/KiCad/PROJECTS/nano-v3/NanoV3.3/attempt-1/NanoV3.3-F_Cu.gbr --outline=/home/user/CAD/KiCad/PROJECTS/nano-v3/NanoV3.3/attempt-1/NanoV3.3-Edge_Cuts.gbr --output-dir=/home/user/CAD/KiCad/PROJECTS/nano-v3/pcb2gcode-out --metric=true --metricoutput=true --mirror-axis=0.0000 --mirror-yaxis=false --nog64=true --optimise=0.0100 --tile-x=1 --tile-y=1 --tolerance=0.0100 --zchange=20.0000 --zchange-absolute=false --zero-start=false --zsafe=2.0000 --extra-passes=2 --invert-gerbers=false --mill-feed=150 --mill-speed=10000 --offset=0.0500 --voronoi=false --zwork=-0.0620 --drill-feed=150 --drill-side=auto --drill-speed=10000 --milldrill=false --nog81=true --nog91-1=true --onedrill=false --zdrill=-3.5000 --bridges=0.4000 --bridgesnum=8 --cut-feed=30 --cut-infeed=1.1000 --cut-side=auto --cut-speed=10000 --cut-vertfeed=60 --cutter-diameter=1.5000 --fill-outline=true --zbridges=0.0000 --zcut=-2.2000

## What happened:
On running pcb2gcode, It gave me the error: 
Processing input files... Internal error in gerberimporter
POLYGON((-3.93701e-07 -3.93701e-07,-3.93701e-07 3.93701e-07,3.93701e-07 3.93701e-07,3.93701e-07 -3.93701e-07,-3.93701e-07 -3.93701e-07))
MULTILINESTRING()
std::exception

## What did you expect to happen:
The usual gerber to nc conversion.
 
## pcb2gcode version (run pcb2gcode --version to see this):
2.5.0
Git commit: latest-dirty
Boost: 107400
Gerbv: 
Geos: Not installed

Please attach your input files and relevant output files and images.  **Don't forget to include your millproject file and gerbers!**  You might need to zip them.  If you don't provide a millproject file, make sure that you copy-paste the command-line that you used.

[attempt-1.zip](https://github.com/user-attachments/files/20788969/attempt-1.zip)

eyal0, 2025-06-18:
I have a fix for this issue.

May I use your back layer gerber in the test suite?  It will be added to the code repo.

Terak8510a, 2025-06-18:
Its fine with me, but it's not my work, it was imported into kicad from an old eagle arduino-nano project.

PR #721 (2025-06-18): fix: Handle degenerate rings in gerberimporter.
It's possible for a ring to have no area.

This fixes #719.
commit 17c92478 (2025-06-18): fix: Handle degenerate rings in gerberimporter.

It's possible for a ring to have no area.

This fixes #719.
closed 2025-06-18 by eyal0